libopencm3/libopencm3 #1543: STM32L476-Discovery UF2 MSC bootloader dos not enumerate

pavavra, 2024-05-06:
I ported https://github.com/mmoskal/uf2-stm32f103 to STM32L1xx in the past without any problem.

Now I try the same with STM32L476VGT6 on STM32L476-Discovery board, but it will never enumerate and ends up with error code 43 in windows (A request for the USB device descriptor failed.) approximately 15-20 seconds after cable plug-in.

I use otgfs_usb_driver, I have SB24 connected in order to connect VBUS to PA9, I tried various patches to ignore VBUS sensing etc, but still with no luck.

I even tried to replace stm32f107_usbd_init code with initialization code taken from CubeMX generated MSC class device which runs and enumerates correctly, but same with no luck.

I have tried USB clock configuration from MSI, PLLQ and PLLSAI1Q too, but all is the same, I thing, I have it set up correctly.

Any help would be very appreciated. Thank you

gw826943555, 2024-05-10:
Try this  [PR.](https://github.com/libopencm3/libopencm3/pull/1261) I tested on my stm32h750 board and cdcacm example works

pavavra, 2024-05-14:
Thank you, I tryed it all but no luck, still the same problem. I suspect USB clock configuration part, is this config correct?

// USB clock from 48MHz MSI clock
void target_clock_setup(void) {
  rcc_periph_clock_enable(RCC_SYSCFG);

  /* Temporarily enable internal high-speed oscillator (HSI). */
  rcc_osc_on(RCC_HSI16);
  rcc_wait_for_osc_ready(RCC_HSI16);

  /* Select HSI as SYSCLK source. */
  rcc_set_sysclk_source(RCC_CFGR_SW_HSI16);

  /* Set the VOS scale mode */
  rcc_periph_clock_enable(RCC_PWR);
  pwr_set_vos_scale(PWR_SCALE1);

  /* Enable external low-speed (LSE) clock for auto-trimming MSI clock. */
  pwr_disable_backup_domain_write_protect();
  rcc_osc_on(RCC_LSE);
  rcc_wait_for_osc_ready(RCC_LSE);
  pwr_enable_backup_domain_write_protect();

  /* Enable multispeed internal oscillator (MSI). */
  rcc_osc_on(RCC_MSI);
  rcc_wait_for_osc_ready(RCC_MSI);

  /* Increase MSI clock to 48 MHz. */
  rcc_set_msi_range(RCC_CR_MSIRANGE_48MHZ);

  /* Set prescalers for AHB, APB1, APB2.*/
  rcc_set_hpre(RCC_CFGR_HPRE_NODIV);
  rcc_set_ppre1(RCC_CFGR_PPRE_NODIV);
  rcc_set_ppre2(RCC_CFGR_PPRE_NODIV);

  /* Disable PLL oscillator before changing its configuration. */
  rcc_osc_off(RCC_PLL);

  /* Configure the main PLL */
  rcc_set_main_pll(RCC_PLLCFGR_PLLSRC_MSI, 3, 10, RCC_PLLCFGR_PLLP_DIV7, RCC_PLLCFGR_PLLQ_DIV2, RCC_PLLCFGR_PLLR_DIV2);

  /* Enable PLL oscillator and wait for it to stabilize. */
  rcc_osc_on(RCC_PLL);
  rcc_wait_for_osc_ready(RCC_PLL);

  /* Slower flash settings for higher clock speed. */
  flash_dcache_enable();
  flash_icache_enable();
  flash_set_ws(FLASH_ACR_LATENCY_4WS);

  /* Select PLL as SYSCLK source. */
  rcc_set_sysclk_source(RCC_CFGR_SW_PLL);

  /* Wait for PLL clock to be selected. */
  rcc_wait_for_sysclk_status(RCC_PLL);

  /* Set the peripheral clock frequencies used. */
  rcc_ahb_frequency  = 80000000;
  rcc_apb1_frequency = 80000000;
  rcc_apb2_frequency = 80000000;

  /* Disable internal high-speed oscillator. */
  rcc_osc_off(RCC_HSI16);

  /* Select MSI as 48MHz clock source */
  rcc_set_clock48_source(RCC_CCIPR_CLK48SEL_MSI);

  /* Enable the MSI clock PLL (derived from LSE) */
  RCC_CR |= RCC_CR_MSIPLLEN;

  /* Enable USB supply */
  PWR_CR2 |= PWR_CR2_USV;
}

or this one

// USB clock from PLL Q from MSI 4MHz
void target_clock_setup(void) {
  rcc_periph_clock_enable(RCC_SYSCFG);
  rcc_periph_clock_enable(RCC_PWR);

  /* Slower flash settings for higher clock speed. */
  flash_dcache_enable();
  flash_icache_enable();
  flash_set_ws(FLASH_ACR_LATENCY_2WS);

  /* Set the VOS scale mode */
  pwr_set_vos_scale(PWR_SCALE1);

  /* Enable multispeed internal oscillator (MSI). */
  rcc_osc_on(RCC_MSI);
  rcc_wait_for_osc_ready(RCC_MSI);

  /* Increase MSI clock to 4 MHz. */
  rcc_set_msi_range(RCC_CR_MSIRANGE_4MHZ);

  RCC_ICSCR &= ~(RCC_ICSCR_MSITRIM_MASK << RCC_ICSCR_MSITRIM_SHIFT);

  /* Enable external low-speed (LSE) clock for auto-trimming MSI clock. */
  pwr_disable_backup_domain_write_protect();
  RCC_BDCR &= ~(RCC_BDCR_LSEDRV_MASK << RCC_BDCR_LSEDRV_SHIFT);
  rcc_osc_on(RCC_LSE);
  rcc_wait_for_osc_ready(RCC_LSE);
  pwr_enable_backup_domain_write_protect();

  /* Enable the MSI clock PLL (derived from LSE) */
  RCC_CR |= RCC_CR_MSIPLLEN;

  /* Disable PLL oscillator before changing its configuration. */
  rcc_osc_off(RCC_PLL);

  /* Configure the main PLL */
  rcc_set_main_pll(RCC_PLLCFGR_PLLSRC_MSI, 1, 48, RCC_PLLCFGR_PLLP_DIV7, RCC_PLLCFGR_PLLQ_DIV4, RCC_PLLCFGR_PLLR_DIV4);
  rcc_pll_output_enable(RCC_PLLCFGR_PLLQEN);
  rcc_pll_output_enable(RCC_PLLCFGR_PLLREN);
  /* Enable PLL oscillator and wait for it to stabilize. */
  rcc_osc_on(RCC_PLL);
  rcc_wait_for_osc_ready(RCC_PLL);

  /* Select PLL as SYSCLK source. */
  rcc_set_sysclk_source(RCC_CFGR_SW_PLL);
  /* Wait for PLL clock to be selected. */
  rcc_wait_for_sysclk_status(RCC_PLL);

  /* Set prescalers for AHB, APB1, APB2.*/
  rcc_set_hpre(RCC_CFGR_HPRE_NODIV);
  rcc_set_ppre1(RCC_CFGR_PPRE_NODIV);
  rcc_set_ppre2(RCC_CFGR_PPRE_NODIV);

  /* Set the peripheral clock frequencies used. */
  rcc_ahb_frequency  = 48000000;
  rcc_apb1_frequency = 48000000;
  rcc_apb2_frequency = 48000000;

  /* Select PLL Q as 48MHz clock source */
  rcc_set_clock48_source(RCC_CCIPR_CLK48SEL_PLL);

  /* Enable USB supply */
  PWR_CR2 |= PWR_CR2_USV;
}

pavavra, 2024-05-14:
Thank you so much, I did what you suggested and finally it started to work. really thousands and thousands of thanks!!!

closed 2024-05-14 by pavavra