libopencm3/libopencm3 #1556: STM32: Timer function timer_clear_flag bug?

alwinl, 2024-08-03:
The timer_clear_flag function (stm32/common/timer_common_all.c:203) sets the timer status register to the inverse of the passed flag(s). 
As per the STM32 Reference Manual various bits are reserved (depending on the exact timer peripheral) and must be kept at reset value (0).
If we pass in a flag, for instance TIM_SR_UIF <code>timer_clear_flag( TIM2, TIM_SR_UIF );</code> the mentioned bits will be set to "1".
So, I think the comment needs to be modified to 
<code>
void timer_clear_flag(uint32_t timer_peripheral, uint32_t flag)
{
	/* All defined bits are rc_w0, reserved bits are 0 */
	TIM_SR(timer_peripheral) &= ~flag;
}
</code>
This will also clear flags that read "0", but they are already cleared anyway.

kernkraftzwerg, 2024-10-10:
That will produce a possible racing condition, because the read-modify-write of &= is not atomic:
1. The instruction reads the TIMx_SR register
2. A timer event occures and a flag is set
3. The instruction writes the result of the AND operation to TIMx_SR and accidentaly clears the new flag.

alwinl, 2025-02-22:
Fair comment. What about this:

```
/*---------------------------------------------------------------------------*/
/** @brief Clear a Status Flag.

@param[in] timer_peripheral Unsigned int32. Timer register address base @ref
tim_reg_base
@param[in] flag Unsigned int32. @ref tim_sr_values. Status register flag.
*/

void timer_clear_flag(uint32_t timer_peripheral, uint32_t flag)
{
	/*
		All defined bits are rc_w0, reserved bits are 0
		
		The reserved bits of the status register are _not_ rc_w0 therefore
		a 1 in a reserved bit position is not allowed.
		We thus need to ensure that all reserved bits are set in or flag
		parameter so they are unset when we take the inverse.
		
		Each type of timer peripheral has more or less reserved bits, so
		we need to ensure that we set the correct bits, depending on the
		peripheral.
		
		peripheral	| mask				
		----------------------------------
		TIM1		| 0b1110000100000000 (0xE100)
		TIM8		| 0b1110000100000000 (0xE100)
		TIM2		| 0b1110000110100000 (0xE1D0)
		TIM3		| 0b1110000110100000 (0xE1D0)
		TIM4		| 0b1110000110100000 (0xE1D0)
		TIM5		| 0b1110000110100000 (0xE1D0)
		TIM9		| 0b1111100110111000 (0xF9B8)
		TIM12		| 0b1111100110111000 (0xF9B8)
		TIM10		| 0b1111110111111100 (0xFDFC)
		TIM11		| 0b1111110111111100 (0xFDFC)
		TIM13		| 0b1111110111111100 (0xFDFC)
		TIM14		| 0b1111110111111100 (0xFDFC)
		TIM6		| 0b1111111111111110 (0xFFFE)
		TIM7		| 0b1111111111111110 (0xFFFE)

	I do not have the documentation for 15 to 21

	 */
	switch( timer_peripheral ) {
	case TIM1 :
#if defined(TIM8)
	case TIM8 :
#endif
		flag |= 0xE100;
		break;
		
	case TIM2:
	case TIM3:
#if defined(TIM4)
	case TIM4:
#endif
	case TIM5:
		flag |= 0xE1D0;
		break;
	
#if defined(TIM9)
	case TIM9:
		flag |= 0xF9B8;
		break;
#endif

#if defined(TIM12)
	case TIM12:
		flag |= 0xF9B8;
		break;
#endif
	
#if defined(TIM10)					
	case TIM10:
		flag |= 0xFDFC;
		break;
#endif
		
#if defined(TIM11)					
	case TIM11:
		flag |= 0xFDFC;
		break;
#endif
		
#if defined(TIM13)					
	case TIM13:
		flag |= 0xFDFC;
		break;
#endif
		
#if defined(TIM14)					
	case TIM14:
		flag |= 0xFDFC;
		break;
#endif
	
	case TIM6:
	case TIM7:
		flag |= 0xFFFE;
		break;
	
	default:
		break;
	}
	
	TIM_SR(timer_peripheral) = ~flag;
}
```