libopencm3/libopencm3 #1560: STM32F3x: rcc_clock_setup_pll hangs when called multiple times

silischulzi, 2024-09-18:
The function `rcc_clock_setup_pll` will hang waiting forever in `rcc_wait_for_osc_not_ready(RCC_PLL)` when called multiple times. Since the PLL is already configured as system clock by a previous call of `rcc_clock_setup_pll`, the PLL can't be disabled and therefore the wait condition will never be met.

The deprecated function `rcc_clock_setup_hsi` had it right, by using the HSI as system clock while reconfiguring the PLL.

Adding those lines at the start of `rcc_clock_setup_pll` should fix the problem:

```
/* Enable internal high-speed oscillator. */
rcc_osc_on(RCC_HSI);
rcc_wait_for_osc_ready(RCC_HSI);
/* Select HSI as SYSCLK source. */
rcc_set_sysclk_source(RCC_CFGR_SW_HSI);
rcc_wait_for_sysclk_status(RCC_HSI);
```

karlp, 2024-09-18:
duplicate of https://github.com/libopencm3/libopencm3/issues/1467  not likely to be worked on in a manner that makes it permanently safe in all call scenarios.

karlp, 2024-09-18:
see also https://github.com/libopencm3/libopencm3/issues/1115 for another case of this.

closed 2024-09-18 by karlp