libopencm3/libopencm3 #1575: ram offset stm32f7 | ld/devices.data:593

seganoff, 2025-02-24:
stm32f7 END ROM_OFF=0x08000000 RAM_OFF=0x20010000 CPU=cortex-m7 FPU=hard-fpv5-sp-d16

i think should be:
stm32f7 END ROM_OFF=0x08000000 RAM_OFF=0x20020000 CPU=cortex-m7 FPU=hard-fpv5-sp-d16

comparing with RM0410 Rev 5, 2.2.2 : SRAM1 starts from 0x2002_0000

seganoff, 2025-02-24:
RAM_OFF=0x20010000
True with rm0385 Rev 9, 2.2.2, cuz less ccm on f75xxx & f74xxx
Also True with rm0431 Rev 4, 1.6.2;             f72xxx & f73xxx

But for f77xxx & f76xxx its 
RAM_OFF=0x20020000

karlp, 2025-02-24:
probably needs more finegrained regexs, it's a common problem, trading off giant tables of exact parts vs attempting to read between the lines and sometimes getting it wrong.

seganoff, 2025-02-26:
```
diff --git a/ld/devices.data b/ld/devices.data
index ed0aefa0..2020aff9 100644
--- a/ld/devices.data
+++ b/ld/devices.data
@@ -179,8 +179,8 @@ stm32f765?g* stm32f7ccm ROM=512K RAM=384K CCM=128K
 stm32f765?i* stm32f7ccm ROM=2048K RAM=384K CCM=128K
 stm32f7[45]6?e* stm32f7ccm ROM=512K RAM=256K CCM=64K
 stm32f7[45]6?g* stm32f7ccm ROM=1024K RAM=256K CCM=64K
-stm32f7[67]7?g* stm32f7ccm ROM=1024K RAM=384K CCM=128K
-stm32f7[67]7?i* stm32f7ccm ROM=2048K RAM=384K CCM=128K
+stm32f7[67]7?g* stm32f7ccm128 ROM=1024K RAM=384K CCM=128K
+stm32f7[67]7?i* stm32f7ccm128 ROM=2048K RAM=384K CCM=128K
 stm32f769?g* stm32f7ccm ROM=1024K RAM=384K CCM=128K
 stm32f7[67][89]?i* stm32f7ccm ROM=2048K RAM=384K CCM=128K
 stm32f750* stm32f7ccm ROM=64K RAM=256K CCM=64K
@@ -553,6 +553,7 @@ nrf52840* nrf52fp ROM=1024K RAM=256K
 stm32f3ccm stm32f3 CCM_OFF=0x10000000
 stm32f4ccm stm32f4 CCM_OFF=0x10000000
 stm32f7ccm stm32f7 CCM_OFF=0x20000000
+stm32f7ccm128 stm32f7_128 CCM_OFF=0x20000000
 stm32g4ccm stm32g4 CCM_OFF=0x10000000
 stm32l1eep stm32l1 EEP_OFF=0x08080000
 
@@ -590,7 +591,9 @@ stm32f2 END ROM_OFF=0x08000000 RAM_OFF=0x20000000 CPU=cortex-m3 FPU=soft
 stm32f3 END ROM_OFF=0x08000000 RAM_OFF=0x20000000 CPU=cortex-m4 FPU=hard-fpv4-sp-d16
 stm32f4 END ROM_OFF=0x08000000 RAM_OFF=0x20000000 CPU=cortex-m4 FPU=hard-fpv4-sp-d16
 #stm32f7 is supported on GCC-arm-embedded 4.8 2014q4
+#stm32f7[67]* END ROM_OFF=0x08000000 RAM_OFF=0x20020000 CPU=cortex-m7 FPU=hard-fpv5-sp-d16
 stm32f7 END ROM_OFF=0x08000000 RAM_OFF=0x20010000 CPU=cortex-m7 FPU=hard-fpv5-sp-d16
+stm32f7_128 END ROM_OFF=0x08000000 RAM_OFF=0x20020000 CPU=cortex-m7 FPU=hard-fpv5-sp-d16
 stm32l0 END ROM_OFF=0x08000000 RAM_OFF=0x20000000 CPU=cortex-m0plus FPU=soft
 stm32l1 END ROM_OFF=0x08000000 RAM_OFF=0x20000000 CPU=cortex-m3 FPU=soft
 stm32l4 END ROM_OFF=0x08000000 RAM_OFF=0x20000000 RAM2_OFF=0x10000000 RAM3_OFF=0x20040000 CPU=cortex-m4 FPU=hard-fpv4-sp-d16
```